LDMX-Software/pflib #20: DMA Readout

tomeichlersmith, 2022-03-04:
@pbutti @slacrherbst Can you copy the instructions here?

pbutti, 2022-03-04:
Hi!
I do not have DMA readout standalone sw. If the instructions were provided in the past, perhaps @slacrherbst  @bengineerd have some info.
If you need I can point you to the line where rogue reads from AxiStreamDma, but I'm sure you have already seen that?

tomeichlersmith, 2022-03-04:
For potentially future reference:
```python
self.febSrpTdests = [i*16 for i in range(12)]
self.febSrpDmaChannels = [rogue.hardware.axi.AxiStreamDma('/dev/axi_stream_dma_0', dest, True) for dest in self.febSrpTdests]
```

tomeichlersmith, 2022-03-07:
Helpful resources from Ryan:
https://github.com/slaclab/aes-stream-drivers/blob/main/rce_stream/app/src/dmaRead.cpp

PR #25 (2022-03-09): DMA Readout and Scriptable pftool
This resolves #23 and resolves #20 with some developments by @jmmans on the jeremy-dev branch. I have rebased that branch onto main and aligned the access patterns that have changed since Jeremy first branched.
closed 2022-03-09 by tomeichlersmith