LDMX-Software/pflib #108: Charge Injections and LED Flashing in pftool

Skelpdar, 2025-05-15:
Sending commands for internal and external charge injections seem to be implemented in the fast control already:
https://github.com/LDMX-Software/pflib/blob/3a94560a4242377bc9a7bbd0990a9b71cc846fb9/src/pflib/FastControlCMS_MMap.cxx#L199 

But, we can not make use of these in pftool just yet. Assuming that these two FC commands work correctly, this is just a matter of implementing some menu options on the user interface side.

Features:
- Add a DAQ->CHARGE_INT menu option (simple)
- Add a DAQ->CHARGE_EXT menu option (simple)
- Add functionality to control LED bias voltages from pftool (Lennart has some bash scripts to do this, so follow his recipe), e.g. a BIAS->LED_BIAS menu option

tomeichlersmith, 2025-05-20:
The code is there to request a charge pulse, but we need to figure out the input parameters for this pulse. Specifically, the time difference between when we send the charge pulse request and when we send a L1A.

We can determine this time difference by recording the "pipeline depth of the ROC" and comparing this value to a fast control parameter that determines this time difference. The relationship between the pipeline depth and this time offset should be pretty stable, so once we scan the time offset parameter while keeping the pipeline depth fixed, we should be able to code that offset into pflib.

From looking at the HGCROC documents I have available to me, I can't figure out what this pipeline depth or the time offset parameters correspond to, but I wanted to keep my notes out in the open in case someone else can add to them.

tomeichlersmith, 2025-05-30:
And to be even more sure, I did a run with 100 events per `FAST_CONTROL.CALIB` setting and it confirms that we see the pulse at a setting of 20.

[image]

verenamh, 2025-06-24:
As Andreas mentioned previously in this issue, we have now also tried testing the parameter_timescan function with more than one event per timepoint, we chose 50 events here. We have so far only done this with the lowrange capacitor and I have now additionally implemented a heatmap plotting option into parameter-time-scan.py using numpy.histogram2d (I will wait with pushing it to github until Andreas' code is approved into the main branch). The normalization was done by slicing the time axis into the different timepoints and normalizing these slices to a scale from 0 to 1. For the x_bins, I chose the amount of timepoints and the y_bins I chose 128 (from 1024 : 8 = 128 for 1024 ADC bins)

Here are a few exemplary plots for the different CALIBs that we ran:

[image]

[image]

[image]

I also combined all of them into one plot to be able to compare the pulses:

[image]