LDMX-Software/pflib #117: ADC and ADC_tm1 discrepancy

Skelpdar, 2025-05-26:
**Describe the bug**
Using the chip default config, I took some pedestal data and expected to see that the entries in both the ADC and ADC_tm1 columns looked similar. Their distributions should look the same when there is no injected charge or LED flash.

Here are the ADC and ADC_tm1 distributions, with entries from all channels. 
 
[image]

[image]

Clearly, the two are different. Now whether this is a bug in the decoding, in the firmware, elinks/data capture, or something that we've forgotten in the chip configuration, I do not know. 

Also there are some funny digital things going on, with the ADC never reaching a value above 63, and the ADC_tm1 having odd behavior in the larger bits. 

**To Reproduce**
Using the firmware from May 23.

tomeichlersmith, 2025-05-27:
There is definitely something wrong with the decoding but I'm not sure what.

```
tom@appa:~/code/ldmx/hgcroc/pflib$ hexdump -v -n 40 -e '1/4 "%08x" "\n"' data/pedestal-2025-05-27-default.raw 
11888811
beef2025
00000068
00280028
00050005
00050005
f00c0c86
03012c02 # <- first real data word
0c601403
08208a02
```
The first data word is `0x3012c02` and this should create a row in the CSV table that looks like
```
*,*,*,*,*,false,false,48,75,-1,2
```
but that isn't found
```
$ denv ./build/pfdecoder -n 1 data/pedestal-2025-05-27-default.raw -o single.csv
(2025-05-27 17:12:24) [pfdecoder]  info: popping 0 event from stream
(2025-05-27 17:12:24) [decoding]  warn: bad event header flag (0b0101 or 0b0010 for first event): 0110
(2025-05-27 17:12:24) [decoding]  warn: bad common mode leading 12 bits (should be all zero): 000000110000
(2025-05-27 17:12:24) [decoding]  warn: bad event header flag (0b0101 or 0b0010 for first event): 0110
(2025-05-27 17:12:24) [decoding]  warn: bad common mode leading 12 bits (should be all zero): 000000110000
$ grep 'false,false,48,75,-1,2' single.csv 
# nothing
```

tomeichlersmith, 2025-05-27:
Actually nevermind, the word I was pointing to is the common mode word. The very next word `0x0c601403` I can manually translate to `false,false,198,5,-1,3` which is present in link 0 channel 0 of the decoded CSV.

So the good news is that I am more confident that this isn't decoding, the bad news is I don't know why this is. As always, this could be an issue with our configuration so I'm going to move towards attempting

tomeichlersmith, 2025-05-28:
I am very confident its not on the software decoding side and so I've been investigating other areas of the chip to see if I can deduce why we are getting such weird values.

The following plots were made from a pedestal run where
- The elinks are well aligned at least according to the idle word testing.
- This was done on a "clean" roc (aka after a `HARDRESET`)
- `mask_toa=0` and `mask_tot=0` for all channels which I would expect to force the TOA to zero (but it doesn't)

[image]
[image]
[image]

After seeing this, I have grown suspicious of how the 32-bit sample word is constructed on the chip. Maybe some parameters need to be tuned so that measured bits are in the right place?
After spending most of my day stumbling around the parameters in the manual, I found `EXTDATA` in the channel parameters which (along with `GLOBAL_ANALOG_0.SELEXTADC = 1`) would allow me to "forced ADC data" within a specific channel. This seemed like a nice way to test the readout pipeline, so I did
`GLOBAL_ANALOG_0.SELEXTADC = 1` and `CH_17.EXTDATA = 689` and then did a pedestal run. Observations:

- All the channels in the link `0` half have data words that are `0` except channel 17. This makes sense because those channels have `EXTDATA=0` (the default).
- The link `1` half is behaving like before.
- Link 0, Channel 17 reads out `adc = 46 adc_tm1 = 268 toa = 518` which I cannot figure out how that is related to `689` (the reason I chose `689` was to see if there was some bit shifting going on).

Full [decode.log](https://github.com/user-attachments/files/20494979/decode.log) for one event with this setup.

Some ideas before I leave for the day:
- Maybe it has to do with the "align buffer"?
- The `Delay` parameters?
- anything else? I am totally lost

tomeichlersmith, 2025-05-29:
The [updated firmware](https://github.com/LDMX-Software/pflib/wiki/Firmware) has resolved this issue. I am now seeing very clean and well behaved pedestals using the same software for decoding I used yesterday.

[image]
[image]
[image]