LDMX-Software/pflib #118: Problems in decoding pedestal data

cmantill, 2025-05-28:
When taking pedestal data (after aligning the elinks and with the ROC set in RUNMODE=1, CH_0_MASK_TOA=1), we see:
```
[ldmxdaq@umn-zcu102-ldmx pflib]$ ./build/pfdecoder data/pedestal_20250528_175900.raw -o  data/pedestal_20250528_175900.csv  -l -1 --headers
(2025-05-28 18:02:12) [pfdecoder] debug: 3.0.0 (v3.0.0-16-g779aad6)
(2025-05-28 18:02:12) [pfdecoder]  info: popping 0 event from stream
(2025-05-28 18:02:12) [decoding] trace: header scan...
(2025-05-28 18:02:12) [decoding] trace: 0x11888811
(2025-05-28 18:02:12) [decoding] trace: 0xbeef2025
(2025-05-28 18:02:12) [decoding] trace: found header
(2025-05-28 18:02:12) [decoding] trace: 0x00000008 total len: 8
(2025-05-28 18:02:12) [decoding] trace: 0x00000000 link lengths 
(2025-05-28 18:02:12) [decoding] trace: 0x00010001 link lengths 
(2025-05-28 18:02:12) [decoding] trace: 0x00010001 link lengths 
(2025-05-28 18:02:12) [decoding] trace: link 0 length 0
terminate called after throwing an instance of 'std::runtime_error'
  what():  0.QLinkFrame provided data words of incorrect length 
Aborted (core dumped)
```

Adding hexdump:
```
[ldmxdaq@umn-zcu102-ldmx pflib]$  hexdump -v -n 40 -e '1/4 "%08x" "\n"' data/pedestal_20250528_175900.raw
11888811
beef2025
00000008
00000000
00010001
00010001
30000000
30000001
30000002
30000003
````

**Environment (please complete the following information):**
 - AlmaLinux 9 ZCU102 UVA
 - gcc 11.4.1
 - uHAL v2.8.1 g
 - git commit`779aad6e583a3f99719301e86b0a04f66b6f8248`

tomeichlersmith, 2025-05-29:
So, there are two issues here.

First, the exception is not being caught which is poor form and leading to a hard-to-read error message.

Second, thank you for sharing the hexdump because that reveals a problem. I'll manually translate the words
```
11888811 # signal word added by pflib
beef2025 # signal word added by pflib
00000008 # total event packet length added by pflib
00000000 # daq link lengths added by pflib
00010001 # trigger link lengths added by pflib
00010001 # trigger link lengths added by pflib
30000000 # trigger header added by pflib
30000001 # trigger header added by pflib
30000002 # trigger header added by pflib
30000003 # trigger header added by pflib
```
Notice that all of these words are words that are manually added by pflib.
- I would go into the `DAQ.DEBUG`  menu and make sure you can `SW_L1A -> ESPY -> ADV`. Check the event occupancy with `STATUS` in that menu to see that it goes up by one when you `SW_L1A` and down when you `ADV`.
- I would also only use the `DAQ.SETUP.FORMAT = 1`, I have not updated the decoding to handle the output of the ECOND emulation as of yet.

tomeichlersmith, 2025-05-29:
Also also, please configure `git` so that the tags are fetched whenever you `git pull`.[^1] This makes the version built in a bit more understandable. The commit you are on is actually just one commit after the latest release:
```
tom@appa:~/code/ldmx/hgcroc/pflib$ git describe --tags 779aad6
v3.1.3-1-g779aad6e
```

[^1]: I'm not sure how to do this to be honest. It looks like its the default behavior of newer versions of Git (2.43.5 is the one in the ZCU image). You can manually do it with `git fetch --tags`, but that's annoying to do all the time.

closed 2025-06-02 by cmantill