LDMX-Software/pflib #145: SiPM/LED Bias Control

Skelpdar, 2025-06-11:
Let's bring back the Bias class (it's currently all commented out), for control of the SiPM/LED voltage bias DACs, as well as the BIAS menu options to read/write to these.

tomeichlersmith, 2025-06-11:
For context, Jeremy was unable to get this working at UMN either. We are also unsure on why the Bias I2C communication is not working the same as the HGCROC I2C communication.

tomeichlersmith, 2025-06-12:
I am unable to run the manual i2ctransfer at UMN. Perhaps I'm missing additional setup that the bash script does?

```
[eichl008@umn-zcu102-ldmx ~]$ i2ctransfer 23 w1@0x10 $(( 0x80 | (1 & 0x07) )) r2
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will send the following messages to device file /dev/i2c-23:
msg 0: addr 0x10, write, len 1, buf 0x81
msg 1: addr 0x10, read, len 2
Continue? [y/N] y
Error: Sending messages failed: Input/output error
```

tomeichlersmith, 2025-06-13:
Are you sure you guys didn't do some other hardware/external setup to get the bias I2C connection operational?

I can use `i2ctransfer` to read registers on bus 24, but not on bus 23 and its been a long time since we started up the test stand.

```
[eichl008@umn-zcu102-ldmx ~]$ i2ctransfer 24 r1@0x24
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will send the following messages to device file /dev/i2c-24:
msg 0: addr 0x24, read, len 1
Continue? [y/N] y
0x00
[eichl008@umn-zcu102-ldmx ~]$ i2ctransfer 24 r1@0x27
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will send the following messages to device file /dev/i2c-24:
msg 0: addr 0x27, read, len 1
Continue? [y/N] y
0x04
[eichl008@umn-zcu102-ldmx ~]$ i2ctransfer 23 r2@0x20
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will send the following messages to device file /dev/i2c-23:
msg 0: addr 0x20, read, len 2
Continue? [y/N] y
Error: Sending messages failed: Input/output error
[eichl008@umn-zcu102-ldmx ~]$ uptime
 16:38:08 up 63 days,  1:35,  2 users,  load average: 0.01, 0.01, 0.00
```