LDMX-Software/pflib #160: TOT measurements

andpet1324, 2025-06-27:
**Is your feature request related to a problem? Please describe.**
We want to be able to measure out TOT values for extending our ADC range. We need to test TOT and TOA parameters to see what they do. What is the best global TOT/TOA threshold? How much does the threshold vary over different channels? How does the TOT look like as a function of time?

To adjust the global TOT and TOA threshold, we set TOT_VREF and TOA_VREF, respectively.

andpet1324, 2025-07-01:
Scanned tot as a function of calib. We now only sample once per BX. 
We see some odd gap where the tot jumps to a linear regime. Any ideas on why we have this gap?

[image]
[image]

andpet1324, 2025-07-09:
I tried a bisectional scan instead of a binary. In the raw data, we no longer get multiple events with the same conditions, so why we had tot_effs larger than 1 was because of the binary search. We still have the same ranges of TOT_VREF where we don't get any tot triggers.

[image]

[image]

cmantill, 2025-10-27:
Hi @andpet1324 
thanks, please use the trim_toa tasks as a guidance for your function.

Three things to follow up:
- Please edit this wiki - https://github.com/LDMX-Software/pflib/wiki/ROC-Calibration#tot-alignment to include step by step what the user should do
- I think that choosing the same channels that were used for aligning `toa_vref` would be OK for a first pass. 
- This procedure will only select `tot_vref` which is a **global parameter right**? i.e. you do not need to do these steps for every channel... Can you start working on the procedure to select `trim_tot` per channel?

cmantill, 2025-10-27:
BTW @andpet1324 @erik-wallin I am attaching here a PNG with the workflow followed by CMS to calibrate: 

[image]