LDMX-Software/pflib #170: DAQRunConsumer that buffers events from latest run

erik-wallin, 2025-07-07:
This was discussed on Slack, but moving the discussion here.

Let's separate two different use-cases:
1. Sometimes you only want to write an event to disk if it fulfills some specific condition, e.g. having a pulse above a certain readout threshold. For this, you could pass a lambda to the RunConsumer that evaluates each event before writing it to disk.
2. If you are implementing a task with some sort of smart scan, e.g. when scanning over CALIB to see where the TOT threshold lies you'd like decrease or increase CALIB dynamically based on the result. Or more generally, if you want to implement something beyond basic grid scans. For this, the user needs to get a handle on the raw event packets themselves. 

I propose that we modify DecodeAndWrite (or make another DAQRunConsumer) that 

1. Buffers every event taken since the last startRun().
2. Has a function std::vector<packing::SingleROCEventPacket>& readBuffer(), that returns the whole buffer to the user.
3. Optionally, only writes an EventPacket to disk if it fulfills some user specified condition.
4. Clears the buffer the next time startRun() is called.

tomeichlersmith, 2025-07-07:
You make good points, I drop my Lambda request 👍

andpet1324, 2025-07-09:
I used another method using the existing implementation. In the writer, I referenced a vector called buffer to which I saved the event packet:
`std::vector<pflib::packing::SingleROCEventPacket> buffer;
pflib::DecodeAndWriteToCSV writer{
...
  [&](std::ofstream& f, const pflib::packing::SingleROCEventPacket& ep) {
...
buffer.push_back(ep);
}
};`

 After doing a daq_run, I then read out and saved the buffer data and cleared the buffer. This seems to work well from what I've seen, but I think it would be nice with a better implementation.

One problem this method has, is that when we have read out the data and calculated some value `val`, I want to save `val` with the writer. But if I reference `val` in the writer, I am saving the value for the current event in the next event, since the value gets calculated _after_ running daq_run.

tomeichlersmith, 2025-07-10:
This DecodeAndWriteToBuffer class looks good and I would like to comment on it. Could you open a PR with _just_ that class addition?

Also, have you connected with @taylorjcolaizzi https://github.com/LDMX-Software/pflib/issues/167 - they are also working on TOA "alignment" (setting TOA thresholds) and so it would be good if y'all worked together on it.

closed 2025-07-15 by tomeichlersmith